panda-official/TimeSwipe #3: EEPROM: Delete external Interfaces

LeilaPiri, 2019-11-07:


begodev, 2019-11-11:
Interface was removed

k41532, 2019-11-15:
@LeilaPiri could you please clarify reopening?

LeilaPiri, 2019-11-20:
> @LeilaPiri could you please clarify reopening?
issue should be reviewed before being closed.

k41532, 2019-11-20:
ok thought 👍 was review :)

@begodev could you please tell commit of removing interface?

i5abell, 2019-11-22:
@begodev I have seen that you have removed the I2C slave for the 8 pin connector and its connection to the EEPROM buffer in the main.cpp. If I see it right, the HAT EEPROM code block is still the same. Just the headline has changed from "R/W EEPROM data from extension plugs" to "Read-only EEPROM data from extension plugs"? Is it still possible for the rPi to write EEPROM data via HAT I2C or have you disabled it elsewhere?

begodev, 2019-11-22:
@k41532: the commit was:

commit 7ae428607b2859ed5fef26b0b6a8e432fefabe47
Author: vnk <vnk@VNKpc.ru>
Date:   Mon Nov 11 16:57:01 2019 +0300

begodev, 2019-11-22:
@i5abell 

> Is it still possible for the rPi to write EEPROM data via HAT I2C or have you disabled it elsewhere?

No, it is impossible since the writing code has been removed also from CSamI2CeepromMaster & CSamI2Cmem. They are both read-only now

k41532, 2019-11-25:
ok, am I right that we 
- only have write access for HAT EEPROM on SAM,
- have I2C available on 8pin and
- have SPI available on 8pin?

begodev, 2019-11-25:
from the current software part (at the master) we have read only access to HAT EEPROM from rPI,
write access on SAM (to fill the data buffer from the external chip)

And no any software interfaces for 8pin (I2C was removed: 2 lines of code are deleted)

closed 2019-11-25 by k41532