panda-official/TimeSwipe #4: Remote Test EEPROM

LeilaPiri, 2019-11-07:


LeilaPiri, 2019-11-11:
OpenOCD installation under the current distributive of ArchLinux is not so trivial as for Raspbian.
Configure script just skips the --enable-bcm2835gpio option for some reason(doesn't find *ARM keyword in a $host_cpu ? ), so manual patching was required to make a correct build...

acolomb, 2019-11-11:
If you find a solution to this, please consider contributing it to the upstream OpenOCD packaging in Arch.  There is already an up-to-date version of openocd available in the package archives, but it's missing the bmc2835gpio interface code as well, unfortunately.  Would be a breeze to just have it available via pacman...

begodev, 2019-11-13:
/dev/i2c-X doesn't appear after loading i2c_dev module by modprobe under a new system.
Is it planned to use eepromutils for programming the EEPROM or I2C will be complitely hidden?

k41532, 2019-11-14:
@begodev I thought of having utils in repo itself, but only having read function implemented in firmware. Write function needs to be added to calibration firmware. Both are hidden for RPi.

k41532, 2019-11-21:
We have a pi online with new Board (connected EEPROM with pullups).

Type `ssh ssh://root@0.tcp.eu.ngrok.io:10089` (root:panda)

I will replace address within next days

@begodev could you test EEPROM utils with that next?

begodev, 2019-11-22:
Ok. Is there also an I2C-0 dev activated? I'll look there...

k41532, 2019-11-22:
Should be available.

begodev, 2019-11-22:
So external HAT interface seems to be working: I see the board via i2c-0 under the Linux and can readout the data. But is the external EEPROM chip is already connected with pullups? Right now its still "arbitration loss"
May I also change a root password?

k41532, 2019-11-23:
External EEPROM should also have pull-ups, but they are added manually. There might be a connection issue. 🤔

You can change pw - but why?

begodev, 2019-11-23:
> External EEPROM should also have pull-ups, but they are added manually. There might be a >connection issue. 🤔

Yes maybe. I think "arbitration lost" in this case just means there is no "high" level or it goes high too slowly.

>You can change pw - but why?

I just thought if some anonymous outside the team reads this and may decide to login,  some armateur young hacker )

begodev, 2019-11-23:
Oh, sorry this time it was address mismatch...

Now I can some communication with a chip at address 0xA0, at least how the bus is going through the corresponding states. I'm continuing testing...

k41532, 2019-11-23:
👍
Did you have a look at eeprom specs?

https://github.com/raspberrypi/hats/blob/master/eeprom-format.md

begodev, 2019-11-24:
Yes, previously I used eeprom utils to create the eeprom image automatically

k41532, 2019-11-26:
@begodev so you have access on both EEPROM? 

from SAM you can access EEPROM (r/w) and HAT EEPROM (r/w)
from RPi you can access HAT EEPROM (r)

if that's right, close issue and open new one for clarifying r/w utils and content (teaser: should follow this: https://github.com/raspberrypi/hats/blob/master/eeprom-format.md

closed 2019-12-09 by LeilaPiri