panda-official/TimeSwipe #10: transfer driver/example to driver/examples

k41532, 2019-11-14:
please transfer dir into examples with name - "DataLogging"

and add some code to main to store data of all sensors in tsv-format just like in old project.

iluxa, 2019-11-19:
PR created https://github.com/panda-official/TimeSwipe/pull/18:
- moved example to driver/examples/DataLogging
- moved sleep 100ms from user code to the driver code
- added example option --dump to dump to stdout all sensors instead of firsts

k41532, 2019-11-19:
Move sleep back into user loop. Or lets discuss another way - just the way we talked about it on channel.

iluxa, 2019-11-20:
Moved sleep back to the user, new PR:
https://github.com/panda-official/TimeSwipe/pull/19

k41532, 2019-11-20:
hey @iluxa beside changes I am missing integration of properly saving data in file ;) Please Check that and do integration.

iluxa, 2019-11-21:
current version let user put `--dump` flag and redirect standard output to the file. I thinks it is a good way to save data to the file, original version has filename hardcoded, should we add new option like 
`--output` for user to save into the file or keep this implementation with redirecting standard output?

k41532, 2019-11-21:
@iluxa thats a very good idea

iluxa, 2019-11-21:
@k41532 what exactly good - leave it as it is or add `--output` flag?

k41532, 2019-11-21:
😁I mean your idea. —output flag

k41532, 2019-11-21:
Let’s also switch config behavior. 

Let’s put chosen input type as selection in there not as option to be chosen by additional flag as flag as of now. 

Let’s work with flag —config therefore. When none is set we use NORM as default with 0 for each offset and 1 for all gains and transmissions. 

@iluxa 
What do you think?

iluxa, 2019-11-21:
yes, then command line will be:
```
./main [--config <filename>] [--input <input_type>] [--output <filename>]
```
`config.json` is default for `--config` option
first input type from config file is default for `--input` option

k41532, 2019-11-21:
perfect

iluxa, 2019-11-24:
added PR https://github.com/panda-official/TimeSwipe/pull/21

closed 2019-11-24 by k41532