panda-official/TimeSwipe #12: Create modern Readme

k41532, 2019-11-14:
[image]

Example attached.

[image]

Who wants to do it?

k41532, 2019-11-14:
[image]

With title like this ... I just think without logo and PANDA :)

k41532, 2019-11-14:
Perhaps we can integrate Subheaders like:

Install
-> link to readme for building drivers for several os

Documentation
-> link to firmware doc
-> link to communication protocol doc
-> link to driver doc

**RPi Examples**
-> link to Rpi examples

**Customizing Firmware**
-> link to building and flashing firmware
-> link to firmware examples

**License**

Other ideas for Main Readme really appreciated. Also about other readme files -> have them in according repo or together in docs dir? 

Please give me feedback on that @iluxa @begodev

PhoGGy, 2020-02-13:
Driver Documentation Generation from Doxygen:
example: https://phoggy.github.io/TimeSwipe/annotated.html
have a look at my fork of TimeSwipe: https://github.com/PhoGGy/TimeSwipe/blob/doxygen/.github/workflows/main.yml
This is a working Github Action to build the Doxygen Documentation from every push and upload it to the gh-pages branch, from where it gets automatically deployed to GitHub.io

closed 2022-02-01 by pmalessa