panda-official/TimeSwipe #9: implementation of a temperature control system for the cooling fan

LeilaPiri, 2019-11-14:
for this purpose use the integrated temperature sensor of the SAM.

i5abell, 2019-11-15:
The ventilator control is routed on pin PA09 (34) of the SAM.

LeilaPiri, 2019-11-26:
- should be implemented using pulse width modulation (PWM), frequency: 1-9 kHz
- 4 steps: 0%, 33%, 66%, 100%

IngoKaiser, 2020-02-09:
@begodev please give me short feedback about current state on that.

begodev, 2020-02-11:
> @begodev please give me short feedback about current state on that.

At the moment a temperature sensor class is implemented for the SAM (Branch Issue 9).
But it seems temperature readout is not really precise - it has markable dispersion.
As I remember +- several degrees.

IngoKaiser, 2020-02-11:
Do you think that's an software issue or precision of temperature sensor?

begodev, 2020-02-11:
> Do you think that's an software issue or precision of temperature sensor?

I think it is because the way it is calculated: it is actually a ratio of 2 measured points. They have to be measured very precisely via SAM's ADC...
I'm trying to find some precision specs in the manual, but it is only written "The accuracy is dependent on the current temperature, and degrades towards extreme conditions."
But no any certain numbers...

I remembered I get +- several degrees, ther was also some initial firmware at XPlained board with temperature measurements it gave near the same, I can recheck.

IngoKaiser, 2020-02-11:
since we just wanna use it for fan control. some degrees won't make a difference. 
So since temp class is implemented just fan controller according @LeilaPiri requirements would need to be implemented right?

begodev, 2020-02-11:
> since we just wanna use it for fan control. some degrees won't make a difference.
> So since temp class is implemented just fan controller according @LeilaPiri requirements would need to be implemented right?

Yes exactly, fan controller with PWM output

closed 2021-02-22 by pmalessa