panda-official/TimeSwipe #14: Change DAC Behavior: 1-4, Overwrite channels with SAM data on DACsw

k41532, 2019-11-15:
Please also give little example in doc on how to write DAC (old DAC1-2) from SAM manually.

begodev, 2019-11-19:
Excuse me, I'd like to ask what does it mean  "to write DAC (old DAC1-2) from SAM manually"?
A source code example how to programm a SAM's DAC? Or how to send a command TO the SAM to change an Analog Output?
Also I'm going to remove DAC0 & DAC1.
And rename DACA-DACD to AOUT1-AOUT4 that they match to hardware manual Pinbelegung_D_SUB_6pin.pdf.

k41532, 2019-11-19:
We just talked about the feature to set DACsw and be able to send chosen data from SAM/Pi on 2 analog outputs. That’s what I meant. Did explanation help?

k41532, 2019-11-21:
Hey @begodev 
I am discussing new naming with @i5abell. 

The way you changed it (AOUT1-4) might be a littlest confusing now. I just added an image to clarify what we have on hardware side. 

So we have an ADC and DAC right at the SAM --> ADC is used to digitalize data for SAM (LED Control) and DAC is to set Analog Output 3-4 manually from SAM. Am I right so far?

Then we have an additional ADC to digitalize sensor data for memory controller. (No control from SAM)

And then we have another DAC - thats for adding a Offset to sensor data.

changing the naming of this last DAC to AOUT is confusing since we only add offset with that (but not only for analog output, also for SAM and memory controller)

Idea: Change AOUT1-4 to DAC1-4, Change AOUT3-4cold, AOUT3-4

Since you have no control on AOUT1-2 anyway (only with setting Offset DAC1-2

Please give me some feedback on that.

begodev, 2019-11-21:
> Since you have no control on AOUT1-2 anyway (only with setting Offset DAC1-2

So it is not connected directly? It is mixed somehow with an input signal amplified? Then of course it is necessary to roll back to the scheme 4DAC + AOUT3/4.  And maybe to create some short hardware manual of the board to be possible to read ?

k41532, 2019-11-21:
Yes, its mixed, please go to scheme 4DAC + AOUT3/4. Where DAC are for setting offset on the input signal (before it gets to the ADCs and Analog output).

begodev, 2019-11-22:
ok

closed 2019-12-02 by begodev