panda-official/TimeSwipe #15: Looks like we are loosing data

k41532, 2019-11-16:
[image]

This is a triangle signal on the data. It looks like I was to slow with lib. We need to get 48k fps. Do we get that? Whats the cpu load like?

Please do some testing.

iluxa, 2019-11-20:
At first we can add data losing check for intto the library
criteria might be if data received later than 500ms after previous data received

How is it possible to measure fps? Is it expecting to receive 48k records per second?

iluxa, 2019-11-21:
Going to resolve https://github.com/panda-official/TimeSwipe/issues/11 before continue testing, it seems data are losing because of real sleep is different from intended

iluxa, 2019-11-21:
After some testing, I found that printing data takes a lot of time and causes losing data.
Even not printing but formatting float pointer numbers.
The issue can be solved by multithread application with lockfree queue.
I get it working without losing data by using [boost::lockfree:spsc_queue](https://www.boost.org/doc/libs/1_66_0/doc/html/boost/lockfree/spsc_queue.html)
We can generate errors counter from the driver side to the user application.

iluxa, 2019-11-25:
PR: https://github.com/panda-official/TimeSwipe/pull/23
driver reads from sensors to its own buffer in own thread
application reads data from driver buffer
application also reads error counter - how many entries have been lost in the buffer
add debian 32bit packed build

iluxa, 2019-11-25:
checked on raspbian - getting 48K fps

k41532, 2019-11-26:
[image]

closed 2019-11-26 by k41532