panda-official/TimeSwipe #17: Semantic Versioning

k41532, 2019-11-18:
Elements
- [ ] Timeswipe Driver
- [ ] Timeswipe Firmware
- [ ] Timeswipe EEPROM
- [ ] Timeswipe Board

Following in principal: https://semver.org

Given a version number MAJOR.MINOR.PATCH, increment the:
1. MAJOR version when you make incompatible API changes,
2. MINOR version when you add functionality in a backwards compatible manner, and
3. PATCH version when you make backwards compatible bug fixes.
Additional labels for pre-release and build metadata are available as extensions to the MAJOR.MINOR.PATCH format.

- We keep Versioning of all Elements synced
- Timeswipe Board Hardware Changes are always MAJOR Changes (so we only need MAJOR Board rev. No.)
- Version 1.0.0 will be the first one for official release - first version with all current functions (see issues) included.
- Meanwhile we will work in 0.X.X 

Questions
- How to implement versioning? Suggestions please.
- Maintainer is responsible for versioning.

pmalessa, 2021-02-22:
@IngoKaiser can this be closed now?

IngoKaiser, 2021-02-22:
@pmalessa done

closed 2021-02-22 by IngoKaiser