panda-official/TimeSwipe #22: Update Driver API

k41532, 2019-11-25:
Timeswipe

- bool init => runs single Set* for according Board… SetInput: - Define defaults; tbd! Integration when EEPROM is implemented

- bool start(std::function<void(std::vector<Record>)> cb) => error and message (vector of Records) as input for callback!

- response setSecondary => I thought about that. Instead of setting DMS, IEPE, NORM... we have severals boards (two atm - one especially for DMS and one especially for IEPE - both have option to measure normalized signal (NORM) as well - we could choose this with this option - so primary would be default.

- response setBridge(int bridge)

… additional Set*

- bool onButton(std::function<void(bool)> cb)

- bool onError(std::function<void(bool)> cb)

- getEEPROM() => tbd: define result type!

- SPI => should be initialized at init() => other communications should build on that
    - bool send(msg)
    - receive() => tbd: define result type!

response could be always (error + msg)? Let's spar on this and get it integrated fast.

iluxa, 2019-11-29:
https://github.com/panda-official/TimeSwipe/pull/28
onError and onButton are implemented

iluxa, 2019-11-30:
added SetSecondary, Init, SetSettngs, GetSettings

IngoKaiser, 2020-02-09:
@iluxa We will get a third node - therefore please change setSecondary to setMode with new inputs (0=IEPE/DMS according to board - can be seen in eeprom, 1=NORM, 2=DIGITAL - we will come to that later)

Each mode will result in a different SPI Message.
Please ping me for message content when this is ready.

Please delete setbridge - we will do that and some others with setsettings,

Also no need for Init - was bad idea 😇

To keep it flexible I also just thought about replacing onButton with onEvent to catch all possible events from arm - the. User should be able to if on any specific event in callback, right? @iluxa

iluxa, 2020-03-12:
@IngoKaiser Can you please explain
setMode - what this function should do? And what is the format?

> Each mode will result in a different SPI Message.

Which SPI messages?

iluxa, 2020-03-15:
Pushed current changes to [branch](https://github.com/panda-official/TimeSwipe/tree/issue_22)
Waiting for Mode implementation and calibration EEPROM atom on firmware side in [issue 69](https://github.com/panda-official/TimeSwipe/issues/69)

begodev, 2020-03-18:
"Mode" AP is added...

IngoKaiser, 2020-03-27:
@iluxa could you please test setting mode?

closed 2020-04-07 by LeilaPiri

iluxa, 2020-04-13:
Merged issue_22 branch into [NewMenu](https://github.com/panda-official/TimeSwipe/tree/NewMenu) branch. Continue development there.