panda-official/TimeSwipe #31: Testing sporadic communication errors returning by a driver function

begodev, 2019-12-04:
@gilyav has reported about sporadic communication errors returning by spi.receive method. This should be tested.

begodev, 2019-12-05:
spi_test is working more than hour on pi.panda.technology (Port 24) Board rev. 2.2 without any errors...

k41532, 2019-12-10:
@iluxa does errors still occur?

iluxa, 2019-12-10:
@k41532 just tested 3 times on 0.tcp.eu.ngrok.io:10274
error appeared after 12, 2, 3 seconds

begodev, 2019-12-11:
The problem that some symbols in the message are sometimes missing.
This came from several reasons:
1) SPI's FIFO has short length.
2) the interrupt routine seems too slow for current speed - should be optimized.
3) communication speed is too high.

IngoKaiser, 2020-02-09:
@iluxa plesse confirm errors are not showing up any more?

iluxa, 2020-02-11:
Last time I tested there were errors (before merge)
Where can I test it now?

IngoKaiser, 2020-02-11:
Please ping @begodev therefore and give feedback when tested.

iluxa, 2020-02-20:
tested on RPi with timeswipeOS
still getting error `protocol error: !obj_not_found!`

begodev, 2020-02-26:
I've built test_spi from the actual driver sources (with reduced speed=50.000Hz). 
It gives no error so far: 10.0.0.12:/root/vnk/test_spi

iluxa, 2020-03-02:
tested test_spi built on current master on Raspbian 32bit and timeswipeOS 64bit
both works without errors for a long time (>10 mins)

closed 2020-03-17 by LeilaPiri