panda-official/TimeSwipe #34: Implement ci/cd for the driver

iluxa, 2019-12-16:
on git tag created new versions should be generated:
driver packages (for debian and arch)
python and nodejs modules

iluxa, 2020-03-12:
Added configuration to githab workflow for [TimeSwipe](https://github.com/panda-official/TimeSwipe) repo([Actions](https://github.com/panda-official/TimeSwipe/actions)) and [TimeSwipe.js](https://github.com/panda-official/TimeSwipe.js) repo([Actions](https://github.com/panda-official/TimeSwipe.js/actions))

On each push to the master branch CI tries to build all packages - the result is displayed on the badge: 
[image]

Information how to make new release added to [Readme.md](https://github.com/panda-official/TimeSwipe/blob/master/driver/Readme.md)

IngoKaiser, 2020-03-20:
@iluxa would you also update Timeswipe.py and add Actions?

iluxa, 2020-03-20:
Yes, will update today

closed 2020-03-24 by IngoKaiser