panda-official/TimeSwipe #35: Make gain and transmission floats

IngoKaiser, 2019-12-17:
Since we have calculated gains and sensor transmissions with decimals, let’s make them float.

IngoKaiser, 2020-02-09:
@iluxa also according to that - should we replace division by multiplication for performance reasons?

iluxa, 2020-02-10:
yes, replacing division with multiplication should increase the performance

iluxa, 2020-03-03:
PR https://github.com/panda-official/TimeSwipe/pull/66
changed gains and transmissions to float, replaced division to multiplication on sensors read
performance looks same, maybe because of compiler optimizations

begodev, 2020-03-09:
Would it be possible to send already pre-calculated (gain x transmission)=mfactor instead of sending them to the convertChunkToRecord separately?
I think it could remove one FMUL instruction

iluxa, 2020-03-10:
replaced gain*transmission with mfactor on https://github.com/panda-official/TimeSwipe/commit/1f1dd9ce2b063a58a8346c1f8353d090b82437d7

closed 2020-03-11 by LeilaPiri