panda-official/TimeSwipe #42: Merge branch button_marker_blinkstart

i5abell, 2020-01-07:
Please review pull request "#38 Button marker blink start", check the code and merge into master, if you agree. If you see problems, collisions or room for improvement, please write it in the review.

begodev, 2020-01-08:
Hello i5 bell, comments are in the pull requests

i5abell, 2020-01-08:
Hey begodev, 
I understand the points. Please keep one of the functions BlinkAtStart or blinkLED and remove the other one. If blinkLED is kept, please adapt it to the behavior of BlinkAtStart (blinking two times in board color). 
I understand what you mean in respect of global variables. Can you please write the view class you mentioned? I think it is a good idea. I thought too, that it would be good, to have all color variables in one place. 
Please merge or integrate the code otherwise then.

I will add comments for visualization variables soon.

begodev, 2020-01-08:
Ok, then I'll think about how to design the class in the best way...

begodev, 2020-01-08:
I've merged 31_39 to a master. So I think we will continue with the View from this point...

i5abell, 2020-01-09:
Ok.

begodev, 2020-01-10:
Now it's in the View

closed 2020-01-13 by i5abell