panda-official/TimeSwipe #45: add sample rate setting

mkh-welsch, 2020-02-03:
the 5-pol analog filter in the hardware is optimized for the 48kHz samples per second. anyways, if the user wants a lower sample rate, the original signal has to be digitally filtered before downsampling the values. this should be implemented in the driver.

this here seems to be a good starting point.
https://github.com/terrygta/SignalResampler

if we resample block-wise, there are artifacts at the start and the beginning due to the math behind resampling and filters, so it needs paddings from original data to overcome this problem. Here you take larger blocks (adding pads), apply resampling, and then cut the additional pad information in the result.

IngoKaiser, 2020-02-09:
@iluxa we should add new method therefore. with resulting fps in `Start` callback then - default is 48k.

iluxa, 2020-02-19:
Initial implementation is on the branch [issue_45](https://github.com/panda-official/TimeSwipe/tree/issue_45)
implementation is based on https://github.com/terrygta/SignalResampler
with padding

tested on RPi without triangular signal - CPU CORE usage is incremented about 10-15%
Future activities: test with triangular signal; memory copiyng optimizations

iluxa, 2020-02-19:
Tests with triangular signal:

[image]
[image]
[image]

iluxa, 2020-02-19:
First without resample
Second with 24K resample
Third with 5K resample

IngoKaiser, 2020-02-21:
@mkh-welsch expected result?

mkh-welsch, 2020-02-22:
[image]
[image]
looks like the instability, we try to avoid with the padding. try to increase the padding size.

mkh-welsch, 2020-02-22:
@iluxa : maybe dynamically increase the padding size with lower resample rate.

mkh-welsch, 2020-02-22:
effect of not using paddings during resampling.
[image]

top: original without resample
middle: resampled without using padding logic
bottom: resampled with padding.

iluxa, 2020-02-24:
https://github.com/panda-official/TimeSwipe/commit/faec13067c993560de034296f4a68270429e02d3
dynamic padding size is implemented

iluxa, 2020-02-25:
https://github.com/panda-official/TimeSwipe/commit/80c31d1f8a7bbbb37539708f7a9c8c471e129264
constant size slices

iluxa, 2020-03-02:
PR for merge
https://github.com/panda-official/TimeSwipe/pull/62

closed 2020-03-02 by iluxa