panda-official/TimeSwipe #48: if timeswipe driver runs on raspberry and other one tries to connect: error message

LeilaPiri, 2020-02-07:


IngoKaiser, 2020-02-09:
Whats meant by this is: Reading from Hardware-Buffer can only be done once at a time (otherwise clocking would cause weird behavior), we should add handling for that. 

We could add this by requesting `EnableADmes` over SPI before sending own one and returning error if this is already 1. Would it work with two SPI instances? Or would that even be another indicator? Ideas please.

iluxa, 2020-02-10:
maybe It is better to add a general-purpose locking system based on lock files into the library (/var/lock)

IngoKaiser, 2020-02-10:
so trying to run lib would result in error when another instance is already using it? Would that be most simple/stable implementation? @begodev any additional thoughts?

iluxa, 2020-02-10:
yes, lock for library instance
This is the most simple implementation because it is a widely used pattern for any type of application

IngoKaiser, 2020-02-10:
@iluxa what would be result trying to open second instance - proper error message?

iluxa, 2020-02-10:
API call for second instance get fails and print some message to stderr

IngoKaiser, 2020-02-10:
so onError would fire - or just global err?

iluxa, 2020-02-11:
After the user application made the first API call to the library (driver) - lock file is created with PID number of the user app
if another app tries to make any API call - library checks that lock file exist and correspond PID is alive then this call fails with global error

IngoKaiser, 2020-02-11:
Ok, let’s give it a try. When first api call is closed lock file will be deleted right?

begodev, 2020-02-11:
> so trying to run lib would result in error when another instance is already using it? Would that be most simple/stable implementation? @begodev any additional thoughts?

Would it be possible just to add a kind of Spin Lock (Mutex, Critical Section) inside the calling methode: then another instance will wait until the first one complete the request?

begodev, 2020-02-11:
Then a number of instances could run simultaneously?

IngoKaiser, 2020-02-11:
Hmm would make it more complicated atm, since what's "completed request" then? 
I guess it would be nice in further steps to have shared memory - so if one is fetching data, other instances could also access that. But no need for now.

begodev, 2020-02-11:
> Hmm would make it more complicated atm, since what's "completed request" then?

I mean an atomic transfer for the board: send a request, wait, return the data received. And the current instance blocks the bus, for example, with a mutex. Of course, all this concerns SPI communication only. Not for other driver's activity.

IngoKaiser, 2020-02-11:
Ok, this would also need to be done for memory controller then. 

Let’s go for global locking for now. @iluxa will that approach also work for instances from py, js, cpp - with prebuilt binaries?

iluxa, 2020-02-11:
> will that approach also work for instances from py, js, cpp - with prebuilt binaries?
Yes, for any instances - will be on the library level
One limitation for this solution read-write filesystem is required, so it will not work on systems without filesystem or read-only filesystem

IngoKaiser, 2020-02-11:
since we're running on linux anyway - this shouldn't be a problem I guess.

iluxa, 2020-03-03:
PR https://github.com/panda-official/TimeSwipe/pull/65

begodev, 2020-03-06:
Couldn't test DataLogging (non static) on Timeswipe because dpkg is missing? (couldn't install the package).
Building DataLogging for Raspbian according to driver's Readme always leads to "cannot execute binary file: Exec format error" (arm-linux-gnueabihf-g++ toolchain was used)

begodev, 2020-03-07:
Has anyone got the same problem? Or it is a problem of my laptop only?

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.4 LTS
Release:	18.04
Codename:	bionic

Linux vnk-VirtualBox 5.3.0-28-generic #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

begodev, 2020-03-07:
On Raspbian it seems pid file lock is working properly

closed 2020-03-10 by LeilaPiri