panda-official/TimeSwipe #49: Add buffer output

IngoKaiser, 2020-02-09:
Its already implemented in our node-red module. Please add method ("SetBufferSize"?!) to set amount of data to be buffered before send to start callback - e.g. 48k would sent buffered data array every second. If defined number is smaller than received chunk - it would be ignored.

iluxa, 2020-02-11:
Added `SetBurstSize` API call according to requirements
PR: https://github.com/panda-official/TimeSwipe/pull/51

iluxa, 2020-02-11:
Not tested, need environment to test

iluxa, 2020-02-11:
Tested on RPi with hardware

begodev, 2020-02-20:
Failed to build the driver on Debian-9 with default settings because of errors during compilation (too old compiler?)
Going to build it on latest Ubuntu...

IngoKaiser, 2020-02-21:
@iluxa what happens when no burstSize is set? 

Expected behavior would be 48k samples per second

iluxa, 2020-02-21:
> what happens when no burstSize is set?
if  SetBurstSize not called, it is default behavior - call ReadCallback as soon as possible

IngoKaiser, 2020-02-21:
sure, you're right

@begodev when checked, please merge to master, @iluxa building driver is easiest with docker for target right? :)

iluxa, 2020-02-21:
yes, the easiest way to build the driver - from docker,
instructions are in driver/Readme.md

begodev, 2020-02-23:
I've tested Data Logging example: it gaves 4x12000 samples in a timespan less than a second (normally 0.2sec, but sometimes up to 0.7 sec).

I found some minor issues in the example itself (not in the driver) in my opinion: 
1. if  config.json is missing the aplication is terminated with a JSON exception
2. the aplication is always finished with "a terminate called without an active exception"
(doesn't wait for threads are finished?)

I think it would be good to install some debugging enviroment on TimeswipeOS to start driver's debugging section remotly like it is done for SAM... (gdbserver and so on).
Then the driver could be futher inspected with a debugger.

iluxa, 2020-02-25:
> 1. if config.json is missing the application is terminated with a JSON exception
>2. the application is always finished with "a terminate called without an active exception"
>(doesn't wait for threads are finished?)

https://github.com/panda-official/TimeSwipe/issues/56 created

closed 2020-02-25 by LeilaPiri