panda-official/TimeSwipe #50: New Menu Structure

i5abell, 2020-02-10:
@begodev: Please implement the new menu structure as shown in the pdf. If there are any questions, please ask. 

[Menu Overview_10_02_2020.pdf](https://github.com/panda-official/TimeSwipe/files/4182391/Menu.Overview_10_02_2020.pdf)

IngoKaiser, 2020-04-07:
[image]

So this could be provided... so gain, bridge, offset (old zero) and mode (was secondary before) will be changed

PhoGGy, 2020-04-07:
I just reviewed the implemented Menu with @IngoKaiser :

- After a Reset blink 2 times short and return back to visualization instead of settings
- After a long click to save the value in one of the menus, go back to settings instead of staying in the menu
- blink 2 times short instead of once long
- on and active LED are hard to distinguish -> Blink active LED instead and keep all other in on state
- in menus with just 2 or 3 options, only set those 2-3 LEDs in on state and the others off, instead of all 4 in on state
- Change reset implementation: Trigger by very long hold (6 sec) in settings screen instead of the white led settings option, blink 2 times short white after reset

Bug: if you hold very long in save (only in blue setting), it blinks long 2 times blue, then lights red 1sec, then back to current setting. The red light is wrong here.
Thanks!

begodev, 2020-04-07:
I've tested menu once again on my board. So:

>After a Reset blink 2 times short and return back to visualization instead of settings

Do you mean board initial reset? As far as I understood from the picture it should go to visualization mode after startup. Or do you mean setting default values as "Reset"?  I cannot reproduce return to visualisation from the "Reset".
So should it go to the "Settings" right after board initial reset?

>After a long click to save the value in one of the menus, go back to settings instead of staying in the menu

What menu was that? 

>blink 2 times short instead of once long after saving

It should blink once for 300ms...It's something strange for me I'll re-flash the firmware...

>Bug: if you hold very long in save (only in blue setting), it blinks long 2 times blue, then lights red 1sec, then back to current setting. The red light is wrong here.

I think you've started finding offset routine: it is blinking LEDs while running and then can show red LEDs on error (offset was not found for a channel)

it is from description:
"LEDs blinking in Menu3 color, while searching
for the values that have to be set for the
demanded offset. Then if the offset was set
successfully, they are active for 1 second in
Menu3 color or red, if an error occurred."

begodev, 2020-04-07:
>on and active LED are hard to distinguish -> Blink active LED instead and keep all other in on state
in menus with just 2 or 3 options, only set those 2-3 LEDs in on state and the others off, instead of all 4 in on state

Ok, this is clear

begodev, 2020-04-07:
>Change reset implementation: Trigger by very long hold (6 sec) in settings screen instead of the white led settings option, blink 2 times short white after reset

For this long click will be returned to press / release (normal click)  mode

begodev, 2020-04-07:
The firmware built from actual sources is re-flashed
Would you please check also the button LED heartbeat?

begodev, 2020-04-07:
The rest changes are coming soon...

iluxa, 2020-04-13:
Implemented events reading in driver, merged changes from [issue_22](https://github.com/panda-official/TimeSwipe/issues/22)

begodev, 2020-04-14:
Menu structure has been changed according to the points above except
"LEDs blinking in Menu3 color, while searching
for the values that have to be set for the
demanded offset. Then if the offset was set
successfully, they are active for 1 second in
Menu3 color **or red, if an error occurred**."

How to show  searching Offset procedure error in a different way?

Data Visualisation based on Moving Average (120-period)  and StdDev is added.

Sensor drop detection has been added assuming the sensor signal goes to zero after unplugging...but sometimes it is not working at my older board...Because zero is also floating here. 
It is hard to find clear and certain creteria for "sensor off" now...

PhoGGy, 2020-04-20:
I checked the menu on 10.0.0.8 and 10.0.0.12. Both seem to have a wrong menu: In RUN Mode, if I long press, it seems to jump to Gain Settings menu instead of settings menu. If I Long Press again, I'm back to RUN mode. Single Click marker works as expected.

PhoGGy, 2020-04-20:
Feedback on newMenu:
- if you long press, the menu is not entered until you release the button. So you have to guess, when to release the button as you have no immediate feedback. This has to be changed, such that it enters the settings menu if you long press, even if you don't release. From settings mode into "menu" mode the same: You hold the button, it enters the menu once Long-Press-threshold is reached. If you still hold the button, after Very-Long-Press-Threshold it enters Reset Mode.
to be continued...

begodev, 2020-04-20:
>This has to be changed, such that it enters the settings menu if you long press, even if you don't >release. From settings mode into "menu" mode the same: You hold the button, it enters the menu >once Long-Press-threshold is reached

That's how it was working before - long click without release before Very-Long Click has been added

>You hold the button, it enters the menu once Long-Press-threshold is reached. If you still hold the >button, after Very-Long-Press-Threshold it enters Reset Mode.

ok, clear. But should then every Very-Long-Press cause reset from any Menu? Because when you press without release, first Long-click will be generated and menu is changed and only then Very-Long-click is be generated?

PhoGGy, 2020-04-24:
only one bug remains:
RUN Mode -> long Click -> settings Mode -> 2x short click to step to blue blinking inside Settings mode -> long click to step inside blue mode.
If I then save the setting with long click, I get 2 long blinks to signal save (not 2 short ones like in the other menus) and then ca. 1sec all red. Then I'm back to settings mode with blue blinking.
If I then long click again (to enter blue mode), I get the same signal again without entering the blue menu: (2 long blinks and 1sec red and back to settings). If I short press in this state, I don't move through settings like I'm supposed to but move through offsets in blue mode.

I hope that's kinda clear. I´ll attach a small video to show the bug.

PhoGGy, 2020-04-27:
Heartbeat while recording is not working. Once I start the record with the dataLogger example, there is no visual feedback

closed 2020-04-27 by PhoGGy