panda-official/TimeSwipe #55: Read EEPROM from the driver and apply related settings

iluxa, 2020-02-21:


IngoKaiser, 2020-02-21:
when the calibration station is ready, an additional calibration atom is created. From this, the appropriate conversion can be taken according to the pid and settings for gain and supply voltage.

so setGain will be dine automatically - user will not have access later. 
so first step would be to read, second to set according setting

iluxa, 2020-02-26:
Read EEPROM implementation: https://github.com/panda-official/TimeSwipe/pull/58
Further instructions needed how to convert EEPROM data to Settings

iluxa, 2020-02-26:
Merged https://github.com/panda-official/TimeSwipe/pull/58 - it also fixes driver build

begodev, 2020-03-09:
bool TimeSwipeEEPROM::Read(std::string& error) is working, but would be good to have the atoms as function's output.

iluxa, 2020-03-10:
>bool TimeSwipeEEPROM::Read(std::string& error) is working, but would be good to have the atoms as function's output.

I don't think the driver (timeswipe.cpp) should think in terms of `CHatAtom*` structures, the idea is to convert information from atoms to settings like gains

@IngoKaiser do we need to set according settings from EEPROM to the driver in the scope of this issue?

IngoKaiser, 2020-03-10:
@iluxa no, lets create new issue with read from EEPROM and set settings for backlog. 

> bool TimeSwipeEEPROM::Read(std::string& error) is working, but would be good to have the atoms as function's output.

@begodev that does not need to be accessible for user. So I guess we will just use it from read to be flexible and reduce workload.

closed 2020-03-10 by IngoKaiser

iluxa, 2020-03-10:
https://github.com/panda-official/TimeSwipe/issues/69