panda-official/TimeSwipe #61: Artifacts at the data begin

iluxa, 2020-02-27:
It happens unrelated to resampling or burst
[image]
[image]

iluxa, 2020-03-04:
Before the fix driver skip only first bundle of data read (usually 64 values)
According to the tests about the first 900 values makes an "artifact"
It is not specified how much data to skip, it seems that depends on the hardware resampler
it must be specified and put to constant `clearFirst`

PR: https://github.com/panda-official/TimeSwipe/pull/67
`clearFirst` set there to 1000

IngoKaiser, 2020-03-24:
this should be a hardware issue

@thomaseichhorn - could you provide access to one of the new boards (0.4.2) for @iluxa to recheck issue there? then we can just discard pr

thomaseichhorn, 2020-03-24:
I can check this here with one of the new boards and try some values of clearfirst... I'll report back asap.

thomaseichhorn, 2020-03-24:
As it turns out, this is a hardware issue which is fixed from board versions 0.4.2 onwards. Attached is the resulting plot when using the previous default (aka no clearfirst):
[image]

IngoKaiser, 2020-03-24:
@iluxa would you please discard pr #67 - thanks for investigation!

closed 2020-03-24 by IngoKaiser