panda-official/TimeSwipe #63: JSON parser sometimes fails in Data Logging example

begodev, 2020-03-02:
[image]

begodev, 2020-03-02:
Driver & Data Logging Example were built on Debian-10...

iluxa, 2020-03-03:
It seems to be issue with encoding
@begodev can you attach corresponding json file please?

begodev, 2020-03-03:
[config.json.zip](https://github.com/panda-official/TimeSwipe/files/4280886/config.json.zip)

the file is taken from 10.0.0.12/root/vnk

iluxa, 2020-03-03:
Attached file doesn't produce any errors
Maybe this file has been saved in Windows in some encoding like UTF-16, then copied to RPi
It seems encoding detection needed and check if a file exists.

iluxa, 2020-03-05:
Added open file and parse exception handlers: https://github.com/panda-official/TimeSwipe/pull/68

I couldn't reproduce errors from the screenshot: tried run example on RPi several times without any errors.

@begodev Can you please test again and ping me when you get this error again

begodev, 2020-03-06:
Ok, with try/catch blocks added it seems the software is working properly.

closed 2020-03-10 by LeilaPiri