panda-official/TimeSwipe #70: Restructure driver output

IngoKaiser, 2020-03-11:
@iluxa as we just have talked about, please make suggestion for data structure for column (signal)-wise data output.

iluxa, 2020-03-17:
instead of Record data structure used in TimeSwipe atm:
```
struct Record
{
    std::array<float, 4> Sensors{0};
};
```
we can implement class contains bulk sensor data with random access to sensor and data each element:
```
class SensorsData {
    static constexpr size_t SENSORS = 4;
    std::array<std::vector<float>, SENSORS> data;
public:
    const std::vector<float> operator[](size_t num) const;
    size_t SensorsSize() const;
    size_t DataSize() const;    
};
```
so, `SensorsData[i]` is vector of data of sensor `i`

IngoKaiser, 2020-03-17:
I think that's fine @iluxa

iluxa, 2020-03-19:
https://github.com/panda-official/TimeSwipe/pull/71

closed 2020-03-19 by LeilaPiri