panda-official/TimeSwipe #74: Copy current Production firmware and simplify

LeilaPiri, 2020-03-25:
Components to remove:
 

- [ ] Visualization of Measurement Values
- [ ] Menu Control via Button incl. Visualization on LEDs
- [ ]  Blinking after Startup
- [ ]  Setting a marker
- [ ]  Differentiation between Short/Long/Double Click, only single short click needed

PhoGGy, 2020-03-26:
The goal is to create a separate firmware based on the current production firmware. This calibration firmware will be used to test and calibrate the Timeswipe board after production. The mentioned components are not necessary for the calibration firmware.
Please create a branch for that

PhoGGy, 2020-03-26:
Issue 75..82 are for the calibration firmware

begodev, 2020-04-10:
Is it possible to combine #81 & #77 also with master? Because temperature is already  measuring and version/ID should be useful also for production firmware?

PhoGGy, 2020-04-14:
Yes, this is possible

PhoGGy, 2020-04-29:
Once the latest issues are merged into master, please create a CalFW branch and remove listed features

closed 2020-11-04 by PhoGGy