panda-official/TimeSwipe #75: Extend "adjust settings" function

LeilaPiri, 2020-03-25:
- The rPi sends a configuration/settings to the SAM
 - js<{„SetSecondary“ : .., 
     „Gain“ : .., 
	 „DAC1.raw“ : .., 
	 „DAC2.raw“ : .., 
	 „DAC3.raw“ : ..,  
     „DAC4.raw“ : .., 
	 „Voltage“ : .., 
	 „Current“ : .., 
	 „MaxCurrent“ : .., 
	 „DACsw“ : ..
	}
 - Voltage, Current and MaxCurrent option have to be added.
 
**Definition of Done**
- [ ] code for mentioned task is complete. examples & documentations are complete.

- [ ] build chains are tested

- [ ] corresponding branch is closed

- [ ] merging is done

begodev, 2020-04-10:
„Voltage“ : ..,
„Current“ : ..,
„MaxCurrent“ : ..,

what should these parameters manage?

PhoGGy, 2020-04-14:
Current and MaxCurrent Setting will be added on future boards. This can be just added to the SPI Protocol without function.
Voltage is the Output Voltage, which can be set via QSPI and Chip Select CS1 on the Max5715 chip.

PhoGGy, 2020-04-14:
These are changes, which are relevant for production firmware, not only the calibration firmware

PhoGGy, 2020-04-24:
Code checked, looks okay for me

PhoGGy, 2020-04-27:
Checked the voltage setting with the SPI terminal:
[image]
The output needs to be rounded to XX.XX, so 2 digit after decimal point. If I set 10.01, I want to receive 10.01 in the response. If I set 10.025, I want to receive 10.03 in the response. If I set 10, response has to be 10.00.

begodev, 2020-04-29:
What I found is an https://github.com/nlohmann/json/issues/677:

>Is there a way to control the precision of serialized floating point numbers?

nlohman:
> There is no parameter for this, and I doubt that adding one would be beneficial for a lot of users.
> What you could do is to edit function dump_float()

It is a recomendation from nlohman.

It seems that some JSON engine fixes are  required...

begodev, 2020-04-30:
NLOHMAN JSON has been patched with std::sprintf.
It doesn't look really nice, but...
Atm is the fastest solution.
branch Issue_75, firmware flashed @10.0.0.8

PhoGGy, 2020-05-07:
review done, works great :)

closed 2020-05-07 by PhoGGy