panda-official/TimeSwipe #78: Check Calibration Status

LeilaPiri, 2020-03-25:
- rPi sends a Calibration Status Request: js>["CalStatus"]
 - SAM reads Calibration Status from EEPROM and responds with status {"CalStatus": 0/1}
 - Suggestion: Save a known Bit Sequence instead of a single Bit inside EEPROM, e.g. 0xFEED if set

**Definition of Done**
- [ ] code for mentioned task is complete. examples & documentations are complete.

- [ ] build chains are tested

- [ ] corresponding branch is closed

- [ ] merging is done

PhoGGy, 2020-04-16:
This could be merged with #81 as it is just an extra information and needs to be implemented in production firmware as well

PhoGGy, 2020-04-24:
I had a look at the pull request. It seems like getCalStatus is currently just returning 0. That should be read from an EEPROM value and stored in EEPROM. in calibration firmware, this value is read/write, in production firmware its read only

begodev, 2020-04-24:
I thought to put the value inside the calibration atom, but the atom structure is still not defined.
Also it is possible to define Cal Status by the number of atoms, but the number is not protected by CRC, unlike the atom itself (it comes with 16bit CRC)

PhoGGy, 2020-04-27:
Okay, so we will add the Calibration Status together with the calibration data once the atom is defined

closed 2020-04-27 by PhoGGy