panda-official/TimeSwipe #80: Save Calibration Data

LeilaPiri, 2020-03-25:
- rPi sends calibration data: js<{"CalData" : ...}
 - not defined yet
 - SAM saves data to EEPROM, reads it back and compares and responds if successful or not
 
**Definition of Done**
- [ ] code for mentioned task is complete. examples & documentations are complete.

- [ ] build chains are tested

- [ ] corresponding branch is closed

- [ ] merging is done

pmalessa, 2020-12-08:
Please have a look at the new Calibration Atom Specification in the docs folder of the EEPROM_doc branch.
Are you able to implement those JSON Commands in the Calibration Firmware? Do you have any suggestions for improvement?

begodev, 2020-12-13:
I like it

begodev, 2020-12-13:
My suggestions... 
Initially I was thinking to put all atoms into the root dir that it would be possible to change any atom by the specified command, I mean not only cAtom but VendorInfo or GPIO atoms as well and maybe to acreate a separate command like ja< [number of the atom], [data].
But it's only an alternative opinion, I'm not sure now it will be an improvement to existing conception.
Both are possible to implement.

pmalessa, 2020-12-14:
Okay. I would keep it as js<, ja< would just make the communication protocol more complicated. Then please implement it the way I specified it. :)

begodev, 2020-12-17:
ok

dmitigr, 2021-02-15:
Available in master since c160a1cddb068a2179491885ab8344d81cf09cff.

closed 2021-02-15 by dmitigr