panda-official/TimeSwipe #81: Get Info

LeilaPiri, 2020-03-25:
- rPi sends command js>{ "ARMID", "fwVersion"}
 - SAM responds with Hardware Chip ID and firmware Version

**Definition of Done**
- [ ] code for mentioned task is complete. examples & documentations are complete.

- [ ] build chains are tested

- [ ] corresponding branch is closed

- [ ] merging is done

IngoKaiser, 2020-04-19:
@begodev implemented in fw? no need to implement in driver as method  - can be requested by implemented getSetting?! @iluxa

begodev, 2020-04-19:
> @begodev implemented in fw? no need to implement in driver as method - can be requested by implemented getSetting?! @iluxa

Yes, it is in the firmware.  If no any double-checking by another guy is required   (since  I've already checked it). I think it can be merged...

begodev, 2020-04-19:
should be accessible via getSettings ("js>...") from the driver. js> works from the terminal

PhoGGy, 2020-04-24:
code checked, looks okay for me

IngoKaiser, 2020-04-24:
tested on 10.0.0.12:

```
# ./rPIspi 0
SPI-0 Master
type the commands:
->
ARMID>\n
84BCA7B7-53385333-34202020-FF06212C
```

closed 2020-04-24 by IngoKaiser