panda-official/TimeSwipe #85: DMSBoard: Implement Shift Register Abstraction Layer

PhoGGy, 2020-04-07:


begodev, 2020-05-11:
I'd like to ask for what IEPE 1-IEPE 4 stand for? Do they have any relation with previous Gain pins?

begodev, 2020-05-11:
Also about QSPI CS0, QSPI CS1 - do they work the same way like SPI Ch0, SPI Ch 1,mean addressing chip 1-chip 4?

PhoGGy, 2020-05-12:
IEPE1-4 enables the U_Bridge for each channel separately. The same functionality as was previously done by PC07 on SAM can now be done via the shift register for each Input channel.

PhoGGy, 2020-05-12:
The QSPI CS Pins work differently. They are directly connected to the CS Pins of the 2 chips. Therefore, you have to make sure to enable only one at a time

PhoGGy, 2020-06-30:
Wheres the difference between issue_85_86 and issue_85_86_ref branch?
Is the doc updated? Couldn't find it. Thanks!

begodev, 2020-07-26:
The main difference is  some refactoring ( issue_85_86_ref) was made for nodeControl and working with separate measurement channels.
The documentation is completed in my opinion if I didn't miss something

closed 2020-09-21 by PhoGGy