panda-official/TimeSwipe #110: SPI settings are overwritten in the first board run

thomaseichhorn, 2020-09-16:
After booting the Pi and the TimeSwipe board, the first run of the datalogging example via `sudo ./main --config ../config.json --input 1 --trace-spi` (or any other program for that matter) will yield the following output:

```
spi: sent: "js<{ "CH1.gain" : 1 }
"
spi: received: "{"CH1.gain":1.00}
"
EEPROM read failed: "Can not access i2c subsystem. Check drivers are properly loaded"
spi: sent: "Mode<0
"
spi: receive failed
spi: sent: "EnableADmes<0
"
spi: received: "0
"
spi: sent: "EnableADmes<1
"
spi: received: "1
"
spi: sent: "je>
"
spi: received: "{"Bridge":false,"Gain":4,"Mode":0}
"
Gain event: 4
Mode event: 0
spi: sent: "je>
"
spi: received: "
"
spi: sent: "je>
"
spi: received: "
...
```

Note the line:
```
spi: received: "{"Bridge":false,"Gain":4,"Mode":0}
```

This will overwrite any settings the user has set beforehand, either via SPI or e.g. via tswipe.SetSettings.
Running the exact same command again gives:

```
sudo ./main --config ../config.json --input 1 --trace-spi
spi: sent: "js<{ "CH1.gain" : 1 }
"
spi: received: "{"CH1.gain":1.00}
"
EEPROM read failed: "Can not access i2c subsystem. Check drivers are properly loaded"
spi: sent: "Mode<0
"
spi: received: "0
"
spi: sent: "EnableADmes<0
"
spi: received: "0
"
spi: sent: "EnableADmes<1
"
spi: received: "1
"
spi: sent: "je>
"
spi: received: "{"Mode":0}
"
Mode event: 0
spi: sent: "je>
"
spi: received: "
"
spi: sent: "je>
"
spi: received: "
...
```

The line in question is now gone, so user-made settings are kept. Is this behaviour to be expected? If so, why, and how can a user ensure that the first run after boot will use custom settings?

begodev, 2020-10-01:
At startup, the firmware reads the user settings from eeprom and since these settings have been just loaded from the eeprom generates an event.
When Data Logging reads this event at the first time the flag is reseted and it means there is no more events to read.
So at the second start of the Data Logging since the flag is not set no event string is displayed

dmitigr, 2021-06-17:
I don't know why (yet) but TimeSwipe board must be restarted _after_ GPIO pins initialization in order to retrieve measurement data. This means, that firmware is restarted too and RAM (where the board settings resides) resets too. I don't know yet how its possible to avoid the firmware restart to make the driver works, but for now I refactored the driver to reset the firmware upon the very first driver initialization.
Fixed in 373924fbf3806f3e235aab131dda98351fdc9be6.

closed 2021-06-17 by dmitigr